sinara-hw/Fast_Servo #1: [RFC] New EEM: Fast Kasli Servo

dtcallcock, 2019-12-02:
A fresh issue for one of the two boards proposed in [[RFC] Sinara Servo sinara-hw/meta#16](https://github.com/sinara-hw/meta/issues/16).

This would be a high performance board using the FPGA on a Kasli to run the feedback loop.  Board would have ADCs, DACs and all the required analog electronics.

- Roughly equivalent capabilities to the [NIST servo] (https://arxiv.org/abs/1508.06319) (100MSPS parallel DACs/ADCs, few 100ns latency).
- Able to do direct digital modulation/demodulation for things like PDH locks.
- Would need quite a few EEMs. 
- Need to consider timing issues if using ribbon cables.  Could be made as a daughterboard that stacks onto Kasli and uses the EEMs as board to board connectors to avoid this but that would limit us to 1 per Kasli.
- High cost, approaching $1k/ch

gkasprow, 2019-12-02:
There is also [ open source](https://github.com/antmicro/zynq-video-board) design that uses Enclustra SO-DIMM modules.

jordens, 2019-12-02:
Yes. My point was just that FMC seems to be designed to offer you front panel real estate, i.e. is more suited to "front ends" while an FPGA module can live somewhere in the back and doesn't need what FMC gives.

gkasprow, 2019-12-02:
I meant FMC carrier card, not FMC module with FPGA:)

gkasprow, 2019-12-02:
Below is Stabilizer with Enclustra module. It fits nicely
[image]

gkasprow, 2019-12-02:
I'm just looking at pinout tables and there are compatible boards with Artix (MarsAX3), Spartan(Mars MX2), Zynq 7(Mars ZX3), Zynq US (Mars XU3) and Cyclone V SoC(Mars MA3). The only potential issue is only 22 LVDS pairs available. Trenz has at least twice more. In the case of dual parallel ADC and DAC, this may be not enough.

hartytp, 2019-12-02:
> The fact that we are trying to do both DC accuracy and stability and very wide bandwidth is a bit worrying to me though.

Indeed. I largely feel this board should be for closed-loop or AC-coupled applications, so DC accuracy/stability should not be necessary.

The only motivation for DC I've come across for this design is the general "it would be great to have a general-purpose board with some fast ADCs/DACs and an FPGA for sinara"...

dhslichter, 2019-12-02:
> "it would be great to have a general-purpose board with some fast ADCs/DACs and an FPGA for sinara"...

Sounds like the original motivation for Sayma 1.0 :stuck_out_tongue_closed_eyes: Beware the feature creep...

gkasprow, 2019-12-02:
Well, we will also have DC-coupled slow ADC and DAC. So in principle, with some effort, one can combine slow DC processing chain  and fast AC processing chain to get wideband operation.

hartytp, 2019-12-02:
Something like a 2 channel pdq with a decent speed ADC would be a really useful addition to Sinara

gkasprow, 2019-12-02:
Any idea what ADCs and DAC would you want for such application?
I have to spend some money in next two weeks so can buy some eval kits or Trenz modules :)

hartytp, 2019-12-02:
Others may feel differently but I’d go for the same dac as pdq / shuttler. And a decent 16 bit (dc precise) 20Hz parallel ADC. Caveat is that I haven’t given this design much thought and don’t have a clear use case in mind for it right now...

gkasprow, 2019-12-02:
I've just ordered        [ TE0710-02-35-2CF](https://shop.trenz-electronic.de/en/TE0710-02-35-2CF-Dual-fast-Ethernet-Artix-Module-with-Xilinx-Artix-7-35T-Speedgrade-2C) and  [TE0712-02-100-2C](https://shop.trenz-electronic.de/en/TE0712-02-100-2C-FPGA-Module-with-Xilinx-Artix-7-XC7A100T-2FGG484C-1-GByte-DDR3-4-x-5-cm) directly from Trenz.
These are nice modules available online from both Trenz and Digikey. Digi is slightly more expensive if you ship them back to EU :)
I already have devkit for DAC used in Shuttler.
This [ADC ](https://pl.mouser.com/datasheet/2/609/AD9461-878087.pdf)looks interesting. Parallel interface and 5bit INL
MUch more expensive but with [1.5bit INL](https://www.analog.com/media/en/technical-documentation/data-sheets/2107fb.pdf). And it uses DDR/CMOS interface.

dhslichter, 2019-12-03:
> Others may feel differently but I’d go for the same dac as pdq / shuttler.

FWIW the tentative Shuttler design plan is to use a <1 LSB DNL/INL 14-bit DAC with dithering to achieve additional bits at DC, with a fallback of using two DAC channels as coarse/fine.  The PDQ DACs are not recommended for new designs, will be EOL soon.  It's challenging to find a high-speed 16-bit DAC with good DC properties and reasonable power/cost.

dhslichter, 2019-12-03:
> And a decent 16 bit (dc precise) 100MSPs parallel ADC

This will be an expensive ADC....

gkasprow, 2019-12-08:
We can copy entire AFE from [this project](https://ohwr.org/project/fmc-adc-100m14b4cha/wikis/home) and only replace ADC with i.e. [LTC 2107](https://www.analog.com/media/en/technical-documentation/data-sheets/2107fb.pdf)
Are there any applications that require DC-coupling of DAC?

gkasprow, 2020-01-03:
My colleague @zdun8 will take care of this project.
Let's close specifications so he could continue.

hartytp, 2020-01-03:
Yeah, it would be interesting to hear what use-cases people might have for this card...

For the use-cases I have in mind, something like an [ AD9262BCPZ-10](https://www.analog.com/media/en/technical-documentation/data-sheets/AD9262.pdf) would be interesting for the input. Not too expensive (~£40/channel in singles); 10MHz analog input BW (640MHz sample rate); 16 bits; decently DC precise; 15dB noise figure; 300mW/channel.

That's fast enough for all the applications I have in mind

hartytp, 2020-01-03:
My chief aim would be something that's "like stabilizer but faster" e.g. for DC-precise feedback loops with a few MHz closed-loop BW. Combining one of those ADCs with a decent >=~75MHz DAC (ideally 16-bit, but 14-bit might also work, DNL of a few etc) would do the job nicely.

hartytp, 2020-01-04:
hmmm...that ADC won't work as the latency is way too high -- almost 1k cycles of the 640MH modulator clock unless I'm misreading the data sheet. Probably no ADC like that will work for this because of latency. :(

hartytp, 2020-01-04:
Anyway, at a high-level, what I'd like to see is:
- 2 channel card that's mechanically compatible with Stabilizer
- latency + BW suitable for 3MHz closed-loop applications (including latency of loop filter on FPGA)
- fixed +-1V input/output range (+-10V range and PGIAs are nice, but not so easy at this BW)
- PoE for stand alone applications
- 1 or 2 EEM connectors routed to the FPGA. Example use case for this is running this card as an EEM driven by Kasli using the FPGA as a SERDES (using the 1.5GBPs/EEM link developed for Fastino/Phaser)
- maybe add an extra high-speed AFE mezzanine connector if there is space and we can find something that works with the current stackup height.

I'd focus on the feedback/fast servo application for this rather than worrying about anything else. Having the option to use it as a fast ADC/DAC EEM adds another nice extra use-case for free.

---

Designing a SDR card is fun, but orthogonal to the fast servo and would probably want a different design (AC-coupled RF front end rather than DC-precise) so best left to a separate project when/if someone has a use-case/specification for it.

I don't personally see any benefit from trying to make this fast enough to do all digital PDH locks now that we have pounder. And, in any case, an all digital PDH card would presumably want a different design (e.g. 3 DACs per ADC, one for the modulation RF output, one for the "fast/laser current lock" output and one for the "slow/PZT lock" output).

gkasprow, 2020-01-04:
This is a sigma-delta modulator so it has several drawbacks. The resolution depends strongly on the effective sampling rate. That's why the bandwidth is just a few MHz.
Another is the latency - it needs a lot of filtering and decimation to get 16bits of resolution.
I used in the past LTC2107 for CCD imaging and I'm not sure we would find better ADC.
Recently I used 4 x fast [ 5MS/s 18bit](https://www.analog.com/media/en/technical-documentation/data-sheets/AD7960.pdf) converts to increase the sampling rate, but it is not straight-forward.
Faster [10MS/s 18bit SAR](https://pl.mouser.com/datasheet/2/609/238618f-1271669.pdf) converters are also available.

hartytp, 2020-01-04:
The [phase metrology] use-case is interesting. It'd be nice to have a 2 channel ADC for this purpose (rather than two single-channel ADCs).

hartytp, 2020-01-04:
@gkasprow I'll leave component choice to you. :)

hartytp, 2020-01-04:
@gkasprow I'll leave component choice to you. :)

Although, do check out the prior art like the [NIST servo](https://arxiv.org/pdf/1508.06319.pdf), the red pitaya etc

hartytp, 2020-01-04:
(e.g. the [ADC](https://www.analog.com/media/en/technical-documentation/data-sheets/219543f.pdf) on the NIST servo seems quite nice)

zdun8, 2020-01-05:
Hello All, 
It's nice to meet you all and I'm happy to help with the project. Let's summarize the specification of the board. 

- ADC we go for LTC2107 with AFE from [FMC-ADC 100M14b4cha](https://ohwr.org/project/fmc-adc-100m14b4cha/wikis/home), yes?

- DAC - not decided yet? As far as I understand we could use the one used in Shuttler project LTC2000?

hartytp, 2020-01-05:
Hi @zdun8 good to meet you!

Assuming we all agree (and since no one has said anything to the contrary yet, I'll assume we do) that the primary goal for this project is to build a digital feedback controller targeting a few MHz close-loop BW, then I think we can close this issue. Let's open separate individual issues for the ADC/DAC choice etc.

Before we settle on the ADC/DAC choice, can you:
- read the whole of this issue as there is some discussion about what features are important (e.g. overall latency)
- read the NIST servo paper linked above and understand what components they chose and why, as well as the performance they achieved
- also look into the red pitaya stuff (@jordens don't know if you have anything to add on this?)
- then compare/contrast the components you've chosen to the other systems so we know why you've chosen them (e.g. may even be worth producing a table with latency, noise, temp co, cost, power consumption, etc, for the various choices)

It's also worth giving some thought to the input/output front ends at this stage. What sample rate and what analog BW will you target? What latency do you expect for the FPGA stuff (LF etc)?

closed 2020-01-05 by hartytp