sinara-hw/Fast_Servo #2: DAC IC selection

zdun8, 2020-01-06:
The goal of this issue is to discuss the selection of the DAC IC. 

The requirements for DAC channel are as follows:
-DAC channel shall have latency + BW suitable for 3MHz closed-loop applications (including latency of loop filter on FPGA)
-DAC channel shall have fixed +-1V input/output range
-DAC channel bandwidth shall be 100 MHz or 125MHz 
-DAC channel shall have 14 bit ENOB

One option is to use LTC2000, which is used in Shuttler project, but it's quite expensive. 
I have listed a similar ones in this spreadsheet: https://docs.google.com/spreadsheets/d/1j0nTGpnnnBx4UXwtZlGYXvdnvVj7PuF5h2WQjEc7hA4/edit?usp=sharing

Feel free to update the spreadsheet.

gkasprow, 2020-01-07:
In case of DACs, please also look at the tempco

hartytp, 2020-01-07:
And again latency will be key...

gkasprow, 2020-01-24:
@hartytp is it critical for your applications to have dual-channel, well-matched DAC?

hartytp, 2020-01-24:
> @hartytp is it critical for your applications to have dual-channel, well-matched DAC?

It's always nice to have matching, but that's an extremely weak preference that can/should be overridden by other factors (e.g. cost/performance/simplicity).

The main case where I'd want matched DACs would be IQ mixers etc, but for that I think we should make a separate SDR design with AC-coupled RF AFE. In any case, Phaser solves all our RF DAC needs for now.

gkasprow, 2020-01-24:
And what about tempco? [These](https://datasheets.maximintegrated.com/en/ds/MAX5898.pdfl) Maxim DACs are low cost and offer 1LSB DNL but would be their  110ppm gain tempo be real killer?

hartytp, 2020-01-25:
> would be their 110ppm gain tempo be real killer?

The primary application for this design is closed-loop, so temp co is irrelevant.

A low temp co can be nice for other secondary uses (e.g. a generic DAC+ADC+FPGA design), but I wouldn't want to compromise the design's main goals over that, so if there are no comparable DACs with low temp co then it's not an issue.

gkasprow, 2020-01-25:
The decision is to either use high-end and expensive LTC2000 or one of low-cost DACs

hartytp, 2020-01-25:
@gkasprow My gut feeling is that we should go for the low-cost DAC.

Before making the final call, @zdun8 could you post a table that compares, the LTC2000, MAX chip, AD9783 (NIST servo) and anything else you think is interesting. Include all the interesting things like cost, power consumption, temp co, INL/DNL, noise, etc, etc. I assume this will be good material for your thesis anyway.

zdun8, 2020-01-25:
I have updated the spreadsheet with the information about some DACs that I've found. 
@gkasprow do you have MAX5898 in mind? In general it seems, that it could do the job if we turn off the interpolation, then the delay is small, and SFDR is very good. One limitation is that the input data is interleaved (for Q and I outputs), but with 500 MSPS I think we are fine. 

I think LTC2000 is very expensive and has only one channel. Comparing AD9783 and MAX5898 they present similar performance. MAX has lower INL and DNL, but it has higher latency approx. 48 ns comparing to 16 ns of the AD for 500 MHz DAC clock. MAX is cheaper 27 USD vs 43 USD. Please check if my analysis is correct, I could only find that AD9783 latency is 7 cycles in the datasheet. 

https://docs.google.com/spreadsheets/d/1j0nTGpnnnBx4UXwtZlGYXvdnvVj7PuF5h2WQjEc7hA4/edit?usp=sharing

gkasprow, 2020-01-25:
We do not need 500MS/s but max 125MS/s.  AD9783 has +/2bit DNL which makes it actually 14bit DAC

hartytp, 2020-01-26:
What's our latency budget?

gkasprow, 2020-01-26:
I was targeting 100ns,
We are anyway limited by analog BW and setting time.

hartytp, 2020-01-26:
> I was targeting 100ns,

For the DAC, or for the whole thing?

> I have updated the spreadsheet with the information about some DACs that I've found.

I think you can write off the AD9125 due to unacceptable latency.

> AD9783 has +/2bit DNL which makes it actually 14bit DAC

That's probably okay for this application. More resolutionis nice but non-critical.

gkasprow, 2020-01-26:
I meant 100ns or so, for entire chain

zdun8, 2020-01-27:
Then we don't have a lot of timing budget. From the DACs that I've analysed AD9783 has the smallest latency (7 cycles). It's been also used in NIST servo. Another option would be MAX5878.

hartytp, 2020-01-27:
> I meant 100ns or so, for entire chain

That doesn't leave a lot for the DAC...I guess we can clock the DAC at a higher rate (but only changing the value every few samples). This feels sufficiently tight that we need a full system model before finalizing this choice. But, I suspect we can rule out anything that's more than, say 10 cycles of latency. At that point the list starts to look rather short...

gkasprow, 2020-01-27:
It does not have to be just 100ns, 150 or even 200ns would be probably fine. THe FPGA also needs a few clock cycles to do even simple algorithm.

hartytp, 2020-01-27:
Sure, I agree we can spare an extre 1-2 clock cycles in the DAC, but a latency of, say, 20 cycles is probably not going to cut it (unless we run the DACs at a very high clock speed). So worth reducing our options to only cover cases with an acceptable latency.

As this is also a critical parameter, I think we'll need a full system model sooner rather than later.

hartytp, 2020-01-28:
> I guess we can clock the DAC at a higher rate (but only changing the value every few samples).

Actually, I suppose that doesn't help. To reach higher sample rates, I suspect we'd need a SERDES in the FPGA which would add additional latency. So, without doing a full simulation including FPGA code that meets timing, I'm not sure we can assume a DAC clock rate above 125MHz...

dhslichter, 2020-02-04:
One option to consider is the AD9117, which we are testing out as a Shuttler candidate DAC.  It runs at 125 MSPS with 4 cycles of latency (32 ns), two channels, 14 bits, <0.5 LSB DNL/INL, relatively low power, low cost.

zdun8, 2020-02-09:
@gkasprow Would 14 bit DAC be sufficient for Fast Servo? Initial requirement was to use at least 16 bit DAC.

zdun8, 2020-02-12:
After discussion with @gkasprow we decided that AD9117 is a valid option for Fast Servo. 
It has 4 cycles of delay at 125 MSPS and provides an autocalibration mechanism that lowers the INL and DNL to +/- 0.6 LSB. @hartytp what do you think about this DAC?

hartytp, 2020-02-13:
That looks ok. What's the preference for this v AD9783? I don't feel strongly, just curious....

gkasprow, 2020-02-13:
It has longer latency, with low cost FPGA one won't be able to drive it with 500MHz anyway, DNL of +/- 2 LSB makes it essentially 14bit DAC, +/-4LSB makes it 13-bit precise

hartytp, 2020-02-13:
Fair enough. I haven't checked the specs in detail but assuming the DC performance, noise, stability, etc of the AD9117  look good then go for it!

dhslichter, 2020-02-13:
FWIW we are currently testing AD9117 to verify datasheet performance and look for any other "gotcha"s.  On the list include DC stability, noise, ability to dither to achieve higher effective bits at DC, measuring tempco, etc.  This will all get put into the Shuttler repository once tests are done.  So far, haven't run across anything that would preclude its use here, especially since the fact that it's part of a closed loop will address slow drifts and tempco.  Definitely monotonic at 14 bits, including major carries, but you have to run the autocalibration routine to get this performance (very simple, no external components required).

zdun8, 2020-02-23:
@gkasprow is programmable gain needed for DAC AFE also?

gkasprow, 2020-02-23:
I don't think so

zdun8, 2020-02-23:
Ok

closed 2021-06-16 by gkasprow