sinara-hw/Fast_Servo #3: ADC IC selection

zdun8, 2020-01-07:
This topic is to discuss ADC IC selection. 

From the initial discussion I saw that there are several options for the ADC IC:
-[AD9461](https://pl.mouser.com/datasheet/2/609/AD9461-878087.pdf)
-[LTC2107](https://www.analog.com/media/en/technical-documentation/data-sheets/2107fb.pdf) 
-[LTC2195](https://www.analog.com/media/en/technical-documentation/data-sheets/219543f.pdf)

I have put their comparison in a spreadsheet: https://docs.google.com/spreadsheets/d/1_hZ97hf2xaO5iiprr2quBoDU6UxvHmOqcvnQKTeb41w/edit?usp=sharing

As I understood the main requirements for the ADC are:
- low latency
- 16 bit
- 100/125 MSPS
- reasonable price

Are there any other?

gkasprow, 2020-01-07:
LTC2195 seems to be the right choice. It gives the lowest cost per channel and 2 channels in 1 IC which is essential for phase diff measurements.

hartytp, 2020-01-07:
NB it's worth doing a latency/phase budget for the system. Take the DAC + ADC latency, simulate the FPGA code to calculate the latency, combine that with the bandwidth of the ADC/DAC AFEs and see how fast we can stably operate this system closed-loop...

zdun8, 2020-01-07:
@hartytp okay, but it will take some time to prepare the simulation. Also I would need some help with the FPGA code, shall it work in the same way as in the NIST servo project, shall I use their design as a reference?

gkasprow, 2020-01-07:
you don't have to write a code to simulate how many clock cycles. You just need a pencil and a paper.

zdun8, 2020-01-07:
Okay, what would be the estimated delay of the FPGA processing?

gkasprow, 2020-01-08:
just count cycles. ADC + ADC serializer, IOSERDES, buffering, DAC DDR interface etc

zdun8, 2020-01-08:
Okay, I'm on it.

hartytp, 2020-01-08:
> you don't have to write a code to simulate how many clock cycles. You just need a pencil and a paper.

Yes, but you do need to make sure your paper design meets timing etc.

For the initial design, making everything as low latency as possible probably makes sense. But eventually we'll need some gateware to flash (presumably this is part of the master's thesis?) so it's worth starting that before finalizing the design.

gkasprow, 2020-01-08:
Writing basic HDL is part of this project. We will do it once schematics are finished.

hartytp, 2020-01-08:
Okay, makes sense. I find that simulating as early in the process as possible is a good way of catching issues early before they become annoying to fix.

gkasprow, 2020-01-08:
It's a must. Nobody is able to read 5k pages of FPGA datasheet :)

hartytp, 2020-01-08:
Indeed!

zdun8, 2020-01-08:
Let's combine the proposed approaches, I'll first prepare a pen-and-paper analysis and then work on a basic simulation, with just an ADC readout, fifo, DAC output, based on hypothetically the best components. What do you think?

hartytp, 2020-01-08:
A pen-and-paper analysis is always a good first step! (I'm always a big fan of back of the envelope sanity checks). That should tell us whether we're in the right ball park...

zdun8, 2020-01-08:
With LTC2195 we are copying NIST servo design, which is good and bad at the same time I feel. The IC is proven, but maybe there is something better already on the market since it was released in 2010.

hartytp, 2020-01-08:
it's funny, that mirrors a conversation I had this morning.

Zotino is the same DAC chip I used at the start of my PhD in 2009. Urukul is the same DDS chip I used then as well. Neither were that new then. The value of this work lies in the integration to make a good overall product, not the fact that the ICs themselves are cutting edge (they aren't). Newer isn't always better...

Anyway, that's by the bye. In the end you have to do a parametric search of the manufacturers and compare the cost, power consumption etc and pick whichever chip will do the best job of matching the spec. If it's a close call then always reuse parts you like from previous designs.

zdun8, 2020-01-08:
I absolutely agree! 
Question: 14 bit ENOB for analog input channel is a must?

gkasprow, 2020-01-08:
ENOB depends on bandwidth. We can take 130MS/s ADC with 10bit ENOB and 16bit SFDR, add some filter and get real 16bit ENOB with lower BW.
That's why people usually use SFDR. Treat ENOB as a wide-band SNR while SFDR as a narrowband SNR.
You probably won't' find 14bit ENOB in ADCs faser than 50MS/s due to thermal noise.

zdun8, 2020-01-08:
Understood, thanks! We need then an ADC with the best SFDR for the low freq BW.

zdun8, 2020-01-10:
I have updated the spreadsheet with SFDR info for low frequency input signals. Please have a look and check if I'm correct. 

I'm not sure how to analyse the SFDR for ADS42LB69, shall we take into account the distortions from 2nd and 3rd harmonics? 

The nice thing about the ADS42LB69 is that it has a DDR interface so the latency will be very low. The not nice thing is the price though...

zdun8, 2020-01-23:
After another discussion with Greg we agreed that the best option would be to use LTC2195, I can't find any better device in this price range and capability. Also it has been proven in Nist Servo project.

hartytp, 2020-01-24:
> After another discussion with Greg we agreed that the best option would be to use LTC2195

Sounds sensible. The NIST servo is pretty well designed AFAICT and design reuse is always a good idea.

zdun8, 2020-02-15:
Which AFE should we go for? The one from NIST servo or FMC ADC 14 bit?

gkasprow, 2020-02-15:
We need to have adjustable gain. I'm not sure about the offset adjustment. For such frequencies, programmable input termination seems to be the must.

hartytp, 2020-02-15:
> We need to have adjustable gain

That's nice but I wouldn't call it a requirement. 

NB we need to consider the phase shift due to the limited AFE BW.

A differential input is also very desirable.

> I'm not sure about the offset adjustment

What kind of circuit did you have in mind?

> For such frequencies, programmable input termination seems to be the must.

Even at a few MHz? If it's easy to add, sure, but I'd worry about parasitics if it's done using the switches on Sampler.

zdun8, 2020-02-23:
So what is the agreement on the requirements for the AFE for ADC for Fast Servo?
Do we need the programmable gain? I'm attaching a circuit for NIST servo AFE, which has a VGA LTC8251. The input can be set to ±0.5 V, ±1 V, ±2 V, or ±4 V . 
[image]

gkasprow, 2020-02-23:
The [FMC-ADC](https://ohwr.org/project/fmc-adc-100m14b4cha/wikis/home) board developed by CERN
https://edms.cern.ch/ui/file/2258647/1/EDA-02063-V6-0_sch.pdf
uses opto-MOS switches which have low parasitic capacitances.

zdun8, 2020-02-23:
Okay, I'll use the FMC-ADC AFE then.

gkasprow, 2020-02-23:
I meant that for switching the input termination you can use same opto-MOS as CERN did. The NIST AFE is much more simple than CERN one. If that is sufficient, I'd go for NIST one.
@hartytp by offset adjustment I meant the circuit the CERN AFE is using. It's a dedicated DAC that adds DC to the input signal.

gkasprow, 2020-02-23:
We are talking about 14bit system. With long cable, even at 10MHz you can have issues with termination.

zdun8, 2020-02-23:
Okay, let's summarize. We go for NIST AFE design and add selecable input termination and offset adjustment that will be controlled with opto-MOSes as in CERN design, yes? Gain adjustment we leave to AD8251.

gkasprow, 2020-02-23:
Termination, Yes. Offset needs discussion. I don't think we need it.

dtcallcock, 2020-02-24:
The NIST design turned out to be very fragile in practice.  A good fraction of boards have had to have their input amps replaced at some point.  Some thought should be put into ruggedizing this somewhat.

hartytp, 2020-02-24:
>  Offset needs discussion. I don't think we need it.

I wasn't assuming we'd have offset subtraction. What's the use case for that in the CERN design?

Generally, that CERN design is quite complex and I wouldn't want to build something like that here unless there is a clearly defined use-case that requires it and so far one hasn't been presented.

Switchable terinations using low capacitance switches seems fine.

> Gain adjustment we leave to AD8251.

The real question for me is what the front-end amp looks like. AFAICT the AD8251 is too slow for this design. I'd love to see a simple model of the closed-loop system that accounts for the various latencies and bandwidths. But, naively I'd assume that the loss of phase margin from that chip is too high. Particularly, bear in mind that this needs to be a DC-precise design so we'd usually want an EMI filter with cut-off below the InAmp BW (to avoid rectification).

If that chip is fast enough then great. If not, can we still have an adjustable gain differential (high CMMR) amplifier? If so, do we end up building our own high-speed PGIA from OpAmps and switches/resistor networks (but then we have to be quite careful about capacitance matching to keep the AC CMMR high). 

> The NIST design turned out to be very fragile in practice. A good fraction of boards have had to have their input amps replaced at some point. Some thought should be put into ruggedizing this somewhat.

Yes. That's something I thought a lot about for Sampler and other designs. So far I'm not aware of any failures on those boards, which is a good sign. But, this is something we'll have to keep in mind if we reduce the input resistors to increase the bandwidth.

zdun8, 2020-02-29:
What bandwidth is required for Fast Servo? In NIST servo paper it says that the input BW was 3 MHz.

hartytp, 2020-03-01:
I’d love to be able to get 3MHz closed loop bw assuming minimal (25deg say) phase shift from external components. Certainly we need more than 1MHzout if this

Would be great to see an analysis of the closed loop system showing the phase lag of the various parts (dac, ADC, loop filter, afe, etc)

dhslichter, 2020-03-03:
> The NIST design turned out to be very fragile in practice. A good fraction of boards have had to have their input amps replaced at some point. Some thought should be put into ruggedizing this somewhat.

Some further study here indicates that the regulators used for these amps were not rated to supply as much current as was required, and thus there was droop on the rail and excessive heating of the regulators.  Apparently the cooling fans also were affected by this rail droop, and this then exacerbated the problem, causing some amps to fail due to overtemp operation.  We have since started replacing the regulators with a pin-compatible model that has a higher current capability.  Anyway, that's the news there -- it seems it may not be the amps' fault that they were blowing.  There are no TVS on the input, though, IIRC, so that might also contribute to some failures potentially.

gkasprow, 2020-03-06:
I don't like the idea of using ad8251. It is slow and thus will define the bandwidth. If we want to have well-matched channels we need to go to something faster where the AFE response can be defined by the external components.
I'd simplify the CERN circuit

gkasprow, 2020-03-06:
The CERN circuit supports 100mV, 1V and 10V range as well as input disconnect and offsets calibration. The last two options are not needed.
The ADA4899 ICs is available as a dual-channel version which will have improved matching

gkasprow, 2020-03-06:
I'd use this circuit
[image]

gkasprow, 2020-03-06:
The only thing I'd change is cutoff frequency

gkasprow, 2020-03-06:
The AD file is [here](https://www.dropbox.com/s/nyfcjq627tcw026/input_stage.SchDoc?dl=0)

hartytp, 2020-03-07:
@gkasprow That approach will have very poor CMMR won't it?

What about a topology more like 
[image]

But, generally, maintaining DC/AC CMMR will be challenging with this kind of design as the BW pushes up.

hartytp, 2020-03-07:
How do those switches compare with the "analog switches" sold by ADI etc?

hartytp, 2020-03-07:
We should aim to keep an EMI filter + protection like Stabilizer.

gkasprow, 2020-03-07:
these switches have no charge injection and have very nearly 0 capacitance to GND since there is no GND terminal. Of course we will add CMC at the input to improve the CMRR.

hartytp, 2020-03-07:
I don’t think you’ll ever get good low frequency cmrr with this topology

Shall we move this discussion to a new issue since it’s no longer about the adc

zdun8, 2020-03-07:
I have created a separate issue for the discussion. I will close this issue.

closed 2020-03-07 by zdun8