sinara-hw/Fast_Servo #4: Analog Inputs AFE design

zdun8, 2020-03-07:
Hello, 

As suggested I'm creating a separate thread for the dicussion about the AFE design for Analog Channels. Current requirements of the AFE are as follows:
AFE shall have:
- adjustable gain 
- high CMRR for low frequencies
- low latency / phase lag
- should allow for 3 MHz loop bandwidth 

AD8251 which was used in NIST servo might be too slow for our design, whereas the CERN design will have low CMRR compared to an instrumential amplifier.

hartytp, 2020-03-07:
NB A really good reference for the basics is https://www.analog.com/media/en/training-seminars/design-handbooks/designers-guide-instrument-amps-complete.pdf

It's not trivial to build a high-impedance differential front end with >>3MHz BW, good CMRR (AC + DC) and switchable gain. Will need thought and simulation (and thinking about all the nasties like Ron temp co etc that can easily trash the stability of the amp)

gkasprow, 2020-03-07:
Why do we need to make it high-impedance?

hartytp, 2020-03-07:
I guess you're right...anything high-impedance will probably be low BW so stabilizer can be used. Still, nice to have a setting which gives 10k or so.

zdun8, 2020-03-07:
What gain values exactly do we require? Is gain 1 a requirement? There is a high bandwidth instrumentation amplifier MAX4445, but it is not unity gain stable, minimum gain is 2, higher gains can be achieved by changing a value of resistor which we could do with opto switches.

hartytp, 2020-03-08:
Let's not talk about gains, but rather input ranges.

IMHO for a faster board like this, a +-1V input range is the only one that's critical.

We could consider a +-10V range, but that might be challenging due to slew rates. If it turns out to be impractical, I'd be okay with dropping or replacing with, say, a +-5V range.

A +-0.1V range (or +-0.25V or whatever is easy to implement) could also be nice.

hartytp, 2020-03-08:
I haven't looked at that MAX chip in detail, but we'd need to look at offsets, noise, temp co, etc in detail to make sure it's up to the job. Otherwise, we can build our own from high-speed OpAmps, with will need to think carefully about CMMR, noise, stability, etc (taking into account all relevant parasitics). That's not so hard, but will take a bit of work. Good learning experience for a masters' project though...

zdun8, 2020-03-08:
Hmm I already have masters degree, I don't need another one :D  I'm helping with the project in my spare time, I think it's really interesting. 

Okay, I will look into the simulation, but it seems to me that it's usually easier to use an IC than to use separate components.

hartytp, 2020-03-08:
> Hmm I already have masters degree, I don't need another one :D 

Sorry! It's hard with these online projects to track who is doing what.

> I'm helping with the project in my spare time, I think it's really interesting.

Cool! 

> Okay, I will look into the simulation, but it seems to me that it's usually easier to use an IC than to use separate components.

Completely agree. It's always better to use an IC than to roll ones own (IC designers can make use of matching tricks that are hard with discrete components). I'm just not aware of any good fast PGIAs available off the shelf.

The other option is to accept that PGIA is hard and just have a unity gain diff amp. It would reduce the flexibility a bit but wouldn't be the end of the world.

zdun8, 2020-03-10:
> IMHO for a faster board like this, a +-1V input range is the only one that's critical.
>
> We could consider a +-10V range, but that might be challenging due to slew rates. If it turns out to be [...]

What is the common mode voltage of the input signal? 0V ?

hartytp, 2020-03-11:
Yea approx 0V just some small amounts of ground nosie

gkasprow, 2020-03-11:
What if we take the standard fully diff amp and add two opto-MOS switched resistors in the FB loop?
Something like [this](http://www.ti.com/lit/ds/symlink/ths770006.pdf) which has  CMRR
[image]

What CMRR are we aiming for? Let's say for 10MHz

jordens, 2020-03-11:
Common mode will typically be differential mode divided by two. Not 0 V.

hartytp, 2020-03-11:
> Common mode will typically be differential mode divided by two. Not 0 V.

hartytp, 2020-03-11:
> What if we take the standard fully diff amp and add two opto-MOS switched resistors in the FB loop?
> Something like this which has CMRR

That could potentially work. We'd need to pay attention about all the usual things like Ron (variations/temp co/matching), capacitance matching, bandwidth, noise, etc.

Out of curiosity, what are the pros/cons of that configuration compared with a "classic" InAmp.

> What CMRR are we aiming for? Let's say for 10MHz

I'm not too worried about the HF CMRR since the CMC + RFI filter should give very good rejection of that. It's more the DC and lower frequencies that we need to watch for.

zdun8, 2020-03-15:
> What if we take the standard fully diff amp and add two opto-MOS switched resistors in the FB loop?
> Something like [this](http://www.ti.com/lit/ds/symlink/ths770006.pdf) which has CMRR

This is a fixed gain fully diff amp, maybe THS4509 or PGA870 would be a better option?

I fear that we might have a significantly lower CMRR due to imperfect feedback resistor matching  with opto-MOS switchable resistors, this can be tricky. What do you think?

gkasprow, 2020-03-15:
True, the opto-MOS have non-calibrated ON resistances. We can use one fully differential amp with 1x gain that gives good CMRR and then another one with programmable gain using opto-MOS which would work as an ADC driver. The PGA870 has 30mV offset which is a lot for 14-bit system

hartytp, 2020-03-15:
Yeah a diff amp and a pgia might be the way to go so long as the diff amp doesn’t add to much noise.

zdun8, 2020-03-29:
Could you have a look at this Tina simulation? I have used THS77006 as an input stage and THS4509 as a PGA witch selectable gain through switches (currently ideal). Strangely, the transient simulation shows some distortion on the output VM3, do you know why this might be?

Few points: the AFE currently supports +/- 100 mV input (switches on), and +/-1 V input (switches on). The signal source needs to be adjusted accordingly. 
https://www.dropbox.com/s/p5dsm860ver8yfn/fast_servo_ths77006_ths4509_tran_v1t.TSC?dl=0

[image]
[image]

hartytp, 2020-03-29:
> Could you have a look at this Tina simulation

Who are you asking?

zdun8, 2020-03-30:
> > Could you have a look at this Tina simulation
> 
> Who are you asking?

Anyone, perhaps you @hartytp  or @gkasprow might have an idea what's going on?  Any help would be highly appreciated.

zdun8, 2020-03-30:
Okay, problem is solved. @gkasprow  suggested that I should run the simulation for a longer period 50 ns, as at the begining there will be a transient state, which will cause distortions. Thanks!

hartytp, 2020-03-30:
Nice! Good catch @gkasprow

zdun8, 2020-04-23:
Finally, I have come up with a draft for the AFE using the initial approach with the instrumentation amplifier. 

TL;DR; Two input voltage ranges were simulated 100 mV and 1 V. CMRR is 56 dB for 0-3 MHz for 1V input signal and 50 dB for 100 mV.
RFI filter is set for 10 MHz for differential signal. 

Attaching schematics and simulation report. I'll be happy to hear any suggestions!
[image]

CMRR plot (signal amplitude 1 V ): 
[image]

Transient plot (signal amplitude 100 mV, frequency 1 MHz):
[image]

AC plot (signal amplitude 1 V):
[image]

More details can be found in the report:
https://docs.google.com/document/d/1smgiGiFC6XPziigHHxrSAG0CFXy-PrKfYR3dXpd6QYk/edit?usp=sharing

Sim file for transient: 
https://www.dropbox.com/s/j2d94vl1exfmxap/fast_servo_tran_v2.asc?dl=0

Sim file for AC
https://www.dropbox.com/s/6mvxo21uil9o3ib/fast_servo_ac_v2.asc?dl=0

Sim file for CMRR 
https://www.dropbox.com/s/0210zgl2mrteqy2/fast_servo_cmrr_v5.asc?dl=0

hartytp, 2020-04-23:
Do you have a noise sim?

For the CMRR and other simulations, are you assuming all components have their nominal values, or did you do a monte-carlo/whatever to look at variance?

zdun8, 2020-04-23:
Adding noise figure measured on the output:
[image]

I have just set the components to 0.1% tolerance for resistors, but I think the standard sim doesn't take that into account. I will prepare a Monte Carlo or a worst case scenario simulation. This is not so straightforward, because I think we should used a matched resistor pairs for the feedback resistors.

dtcallcock, 2020-04-23:
Do you have a sense of likely robustness without TVS or similar?  

As I mentioned, the front-ends die all the time on the NIST boxes.

zdun8, 2020-04-24:
There is no information in the datasheet about any input protection. Also an evaluation board for the ADA4817 opamp doesn't have any.

Maybe we can add a high value limiting resistor and low leakage TVS on the input?

Attaching CMRR simulations with Monte Carlo component variance and worst case scenario. Unfortunately it seems that part value variance degrades the CMRR higher frequencies.  RFI filter capacitors 5% tolerance is causing that. We can increase the input filter bandwidth to accommodate for that. 

1. Monte Carlo part value variance. Resistors' tolerance 0.1%, capacitors' 5%. 100 runs. 
[image]

2. Monte Carlo part value variance. Resistors' tolerance 0.1%, capacitors' 5%. 5 runs. Temperature change sweep 0 -> 80 degrees with 20 degree step. 
[image]

3. Worst case simulation. 100 runs. 
[image]

4. Monte Carlo part value variance. Resistors' tolerance 0.1%, capacitors value fixed 5 runs. 
[image]

gkasprow, 2020-04-24:
We need protection diodes as in Stabilizer. They can affect the CMRR due to non-linear capacitance

zdun8, 2020-04-26:
I have modified the RFI filter a bit. I think we don't need low pass filter on each input, common mode choke does the job of filtering common noise, and mismatch of these capacitors significantly impacts the CMRR.

New AFE:
[image]

Modified filter also results in better CMRR and AC characteristics. Plots with protection diodes for 1V input signal, parts value is modeled with Monte Carlo function, 10 runs. Resistors' tolerance 0.1%, capacitors 5%. 

CMRR:
[image]

Noise:
[image]

AC:
[image]

 If the SPICE model from here https://www.diodes.com/part/view/BAV199 is correct the diodes don't affect the CMRR significantly.

gkasprow, 2020-04-26:
The CMRR can be affected by differences between the diodes. That's why it's better to use quad pack diodes which have similar capacitances. The diode capacitance depends strongly on voltage but we assume the CM voltage is the same.

hartytp, 2020-04-26:
@zdun8 can you put the noise plots on a log log axis please? I can't read the noise floor from the plots (just the flicker noise).

hartytp, 2020-04-26:
How does the Ron temp co/matching affect the gain stability + CMMR? Have you estimated the overall gain stability (I'd just do a back of the envelope on paper, looking at the most sensitive parts). What about the switch temp co? NB you have quite low resistor values, which will accentuate the FET imperfections.

What's the self-heating with a FS input and +-11.5V (or whaterver) supplies? You have quite low resistor values so I can imagine this getting hot.

hartytp, 2020-04-26:
Aah, sorry, forgot we're only supporting +-1V max, so the power consumption is less of an issue.

hartytp, 2020-04-26:
A 4pF differential capacitor is quite small, what do you think the parasitics will be?

hartytp, 2020-04-26:
Is that enough capacitance to drive the ADC? Do you also need a differential capacitor for the ADC?

hartytp, 2020-04-26:
What's the motivation for removing the shunt capacitors to ground on the inputs? Won't they help to improve the AC CMMR?

hartytp, 2020-04-26:
What range of offsets do you expect due to resistor tolerances. IME it's usually worth using resistor networks for the gain/subtraction resistors to avoid annoying input offsets that need to be calibrated out later on.

hartytp, 2020-04-26:
Anyway, nice work!

zdun8, 2020-04-27:
> can you put the noise plots on a log log axis please? I can't read the noise floor from the plots (just the flicker noise).

Noise plot for 1 V input signal:
[image]

Noise plot for 100 mV input signal:
[image]

> How does the Ron temp co/matching affect the gain stability + CMMR? Have you estimated the overall gain stability (I'd just do a back of the envelope on paper, looking at the most sensitive parts). What about the switch temp co?

Not yet. What do you think would be the best way to simulate this?

> A 4pF differential capacitor is quite small, what do you think the parasitics will be?

Not sure. We can increase the value of the capacitor if we decrease the value of series resistors. 

> Is that enough capacitance to drive the ADC? Do you also need a differential capacitor for the ADC?

Which capacitor you mean exactly? Of the filter on the output of the LTC6406? 
[image]

> What's the motivation for removing the shunt capacitors to ground on the inputs? Won't they help to improve the AC CMMR?

According to this app note https://www.analog.com/media/en/training-seminars/tutorials/MT-070.pdf page 4, the common mode choke does the same job so the capacitors can be removed. Also when simulating with Monte Carlo part value variance, the CMRR was heavily affected by the mismatch of those values, but it's also possible that my simulation is wrong. The app note suggests that C9 should be 10xC7, but that's difficult to achieve if we want to have the cut off frequency for common mode at 10 MHz and similar for differential.  Attaching some plots:

Schematic of the RFI filter:
[image]

CMRR with the RFI filter capacitors at fixed values, rest of the compontents with Monte Carlo part value variability. 
[image]

CMRR with all components with Monte Carlo part variability. 
[image]

> What range of offsets do you expect due to resistor tolerances. IME it's usually worth using resistor networks for the gain/subtraction resistors to avoid annoying input offsets that need to be calibrated out later on.

Totally agreed. Not sure about the offsets, we need to be careful about the resistors. 

> Anyway, nice work!

Thanks :)

hartytp, 2020-05-05:
Thanks @zdun8!

> Noise plot for 100 mV input signal:

Nit pick: I assume that's RTO, not RTI. I generally prefer RTI plots but in either case please do specify so we know what to look at. 

It's hard to read the noise plots without a grid (another nit pick, please always add a grid and ideally a marker!). AFAICT, the RTI noise for the +-1V setting is approx 10nV/rtHz. Which is pretty good (e.g. same as the AD8253 on Sampler/Stabilizer). I'm assuming it's also about 10nV/rtHz for the 100mV setting, in which case we're doing pretty well.

> Not yet. What do you think would be the best way to simulate this?

@gkasprow what do you think about simulating the effects of Ron/leakage (tolerance plus temp co) on gain/offset accuracy and CMMR?

> Not sure. We can increase the value of the capacitor if we decrease the value of series resistors.

I'll leave this decision to you and @gkasprow. I don't think the series resistor needs to be that big to protect components. But, no point decreasing if you're not worried about the impact of parasitics here.

> Which capacitor you mean exactly? Of the filter on the output of the LTC6406?

Aah, thanks for clarifying -- I hadn't appreciated that your model includes the input state of the ADC.

Anyway, what I meant was that these ADCs normally need quite a large capacitance to supply the sampling charge. The capacitors on the output filter are only a few pF, which may lead to sampling errors.

> According to this app note https://www.analog.com/media/en/training-seminars/tutorials/MT-070.pdf page 4, the common mode choke does the same job so the capacitors can be removed. Also when [...]

Yes, I agree that it's hard to have the capacitors without also reducing the input bandwidth. If your current design works better then I'd say keep it. Thanks for simulating the different designs, always good to test our assumptions.

gkasprow, 2020-11-23:
@zdun8 don't ADC lines need equalisation?

zdun8, 2020-11-24:
@gkasprow  You mean between pairs? I think on 125 MHz the between-pair length-matching range difference is ~33 mm, but there is no harm in doing so, I can check if there is space to tune them.

gkasprow, 2020-11-24:
Isn't it DDR?
Take into account entire timing budget (Th/Tp), skew and you quickly will end up with much lower value..

zdun8, 2020-11-25:
Sure, I'll double check it.

closed 2021-06-16 by gkasprow