sinara-hw/Fast_Servo #5: Analog outputs AFE design

zdun8, 2020-05-06:
Hello, 

I'm creating a new issue for AFE design for DAC channel. Could you please verify that these requirements are valid?

1. DAC channel shall have latency + BW suitable for 3MHz closed-loop applications (including latency of loop filter on FPGA)
2. DAC channel shall have fixed +-1V input/output range
3. DAC channel bandwidth shall be 100 MHz or 125MHz 14bit
4. DAC channel shall use AD9117 DAC
5. DAC channel load will be 50 Ohms

I would like to use ADA4930 FDA on the output to drive +/-1 V signal. What do you think? 
Thanks for your feedback!

hartytp, 2020-05-07:
> Analog output channel bandwidth shall be 100 MHz or 125MHz 14bit

The bandwidth is already covered by requirement (1) so no need to specify that. 

Other than that, that's about what I had in mind, but I'd feel much more comfortable if other users would chip in. (It's hard to be sure that anyone will actually use this if they don't make themselves heard...)

gkasprow, 2020-11-23:
There is an issue with DAC outputs.
In Stabilizer, the SMA connector case was kept on the GND level.
In Fast Servo it is driven in differential way so we loose compatibility. Coaxial cables are not designed to be driven symetrically - for that purpose twinax cables are used.
Please use only positive output of the amplifier

zdun8, 2020-11-24:
Thanks for the check, I'll fix it.

zdun8, 2020-11-25:
Shall we just terminate the N signal after the common mode inductor with 50 Ohms and connect the negative pins of SMA to GND?

[image]

I think simulations should be rerun for such change. Also, what's the point of using a differential output amplifier if we only need single ended output...

gkasprow, 2020-12-16:
yes, but mak the termination resistor DNP

closed 2021-06-16 by gkasprow