sinara-hw/Fast_Servo #7: Schematics draft - review request

zdun8, 2020-07-28:
Hello All, 

I would like to ask you to review the schematics of Fast Servo. They are in a draft stage, but from the electrical point of view they are almost complete. 

Thanks in advance!

zdun8, 2020-07-31:
There is one known issue. The analog clocks for ADC and DAC need to be externally generated by a dedicated IC.

dtcallcock, 2020-07-31:
Would it be possible to post a pdf?

zdun8, 2020-08-02:
Sure, but please give me some time. Due to the fact that I need to add a clock IC, the power supply has to be modified.

zdun8, 2020-08-06:
I have updated the schematics. Attaching a PDF file. As always the schematics are subject to change during the layout :) [
[Fast_Servo_Sch_Draft_V1.PDF](https://github.com/sinara-hw/Fast_Servo/files/5038604/Fast_Servo_Sch_Draft_V1.PDF)
](url)

closed 2021-06-16 by gkasprow