sinara-hw/Fast_Servo #8: Component placement review

zdun8, 2020-08-10:
Hello, 

I have started working on the layout. Please have a look at the screenshots below showing the component placement. 
[image]

[image]

dtcallcock, 2020-08-11:
Is matching the Stabilizer header placement to allow compatibility with Stabilizer mezzanines possible/desirable?

zdun8, 2020-08-11:
I can check if it's possible. Could you send me the step files for the mezzanines? Or the links to the PCB projects?

dtcallcock, 2020-08-11:
Pounder is an example of such a mezzanine.  Hopefully everything you need is there in the [Pounder repository](https://github.com/sinara-hw/Pounder).

gkasprow, 2020-08-11:
@zdun8 we keep full mechanical and electrical (logic levels)  compatibility with Stabilizer. It will propagate to mezzanine compatibility as well. The only difference is missing 10V range.

zdun8, 2020-08-11:
To be honest, I have completely missed that requirement. I'll update the schematics and PCB accordingly.

zdun8, 2020-09-10:
Please check the updated placement. 
[image]
[image]

zdun8, 2020-09-16:
@gkasprow Yes, local power are only LDOs. 

Basic placement version of the PCB is on the repo.

closed 2021-06-16 by gkasprow