sinara-hw/Fast_Servo #9: design review

gkasprow, 2020-08-12:
- [ ] arrange that all designators and texts do no overlap,  make sure they are not inverted (I know about Altium option)
- [ ] please add 0R option resistor that grounds input connectors
- [ ] add 0R between U1A and IC12 for debugging
- [ ] don't we need ESD protection for SD?
- [ ] did U do the power budget? How much power would LDOs dissipate? Please add a suitable note. use ADI tool for power tree simulation.
- [ ] was power sequencing implemented? One must make sure that no SOM signals are present before SOM is supplied
- [ ] please mark which banks are 3.3 and 1.8V rated
- [ ] why we need both R23 and R6?
- [ ] rename P1V0 -> REF1V0
- [ ] add 0R to J8 - L2, L4, just in case

zdun8, 2020-08-13:
Thanks for the review. I will introduce the changes and will let you know if I have any doubts.

zdun8, 2020-08-31:
> * don't we need ESD protection for SD?
There is a voltage level translator between the SD-Card and SoM, so additional ESD protection is not needed.

> * specify which Trenz modules can be used. Take into account the signalling standard and modules based on Artix, Kintex7 and Zynq

For now, we can support TE0712 (Artix), TE0715-15 (Zynq Artix), TE0715-30 (Zynq Kitex), TE7020 (Zynq Artix), is that okay? @gkasprow 

> * did U do the power budget? How much power would LDOs dissipate? Please add a suitable note. use ADI tool for power tree simulation
[image]

gkasprow, 2020-09-01:
are you sure the level converter has ESD protection?
Support for these modules is fine, just place an info on schematics

zdun8, 2020-09-01:
> are you sure the level converter has ESD protection?

Yes, is that enough? 
[image]

gkasprow, 2020-09-01:
ok, should be fine.

zdun8, 2020-10-25:
Routed PCB is ready for review on the repo. Please let me know if you find anything! 
 
I have checked the pinout in vivado for template projects for TE0720, TE0715-15 and TE0715-30 and the bitstream is properly generated. Archived projects are also available on the repo.

closed 2021-06-16 by gkasprow