sinara-hw/Fast_Servo #11: CLK_100M assignment issue

jmatyas, 2020-11-14:
On TE0712 with Artix, CLK_100M is assigned to a pin (C19) that can only be used as the N side of a differential clock input. 

According to Xilinx 7 Series Clocking User Guide [UG472](https://www.xilinx.com/support/documentation/user_guides/ug472_7Series_Clocking.pdf) p. 31:
[image]

Vivado throws an error during compilation and won't allow past this issue, unless one accepts "non-optimal use of the N side of the clock capable pins" ([here](https://www.xilinx.com/support/answers/67599.html)).

zdun8, 2020-11-14:
Interesting, I didn't know that thanks. 

TE0712 is not supported on FastServo, only TE0720, TE0715-15 and TE0715-30. 

On TE0720 and TE0715 CLK_100M is connected to P capable pin, so it should work.

jmatyas, 2020-11-17:
> TE0712 is not supported on FastServo, only TE0720, TE0715-15 and TE0715-30.

Which module with Artix on board will be supported, then?  I emailed Greg a few days ago and he mentioned that a Trenz module with Artix shall be supported on Fast Servo (that's what the wiki page says, as well).

 Or is it not yet decided which module will be used?

zdun8, 2020-11-17:
Unfortunately I had to remove TE0712 support as it has one bank fixed at 3.3V and then it's impossible to be compatible with TE0720 and TE0715-15/30. 

TE0720 and TE0715 have Zynq with Artix FPGA and they are supported on FastServo. Is that okay for you? 

I will update the wiki with the compatible board list.

jmatyas, 2020-11-18:
It seems to be a sensible solution but I'm not sure about leaving the Artix module out. 

@gkasprow  what do you think about not supporting modules with "clean" Artix at all?

zdun8, 2020-11-18:
I totally understand your concern. Initially I thought that supporting different Trenz module will be straightforward, but unfortunately the devil is in the details and it's not easy to support all kinds of modules. 

I would suggest to leave it as it is right now, so TE0720, TE0715-15/30 are supported. I think it gives plenty of flexibility.

gkasprow, 2020-11-23:
So, for the moment we have only ZynQ devices supported? There is no full ARTIQ support for ZynQ right now.

gkasprow, 2020-11-23:
What is exactly the problem with 3V3 bank support? We need 3V3 bank anyway to keep compatibility with 3V3 levels of existing mezzanines

zdun8, 2020-11-23:
In short: on Artix module TE0712 on JM2 connector some pins have fixed voltage whereas on other  modules they are configurable (set via external pins). 

Please have a look on this spreadsheet: 

https://docs.google.com/spreadsheets/d/1bnEOR_nFoHTlCReCy4KJ_HS_cJoHPDwn3MEk9fibMuM/edit?usp=drivesdk

In order to be able to support Zynq Artix nad Zynq Kintex modules some IOs have to be fixed to some banks to allow 3.3V or 1.8V voltage standard support and signals on JM2 pins for TE0712 are fixed on 3.3V whereas on other modules we can change them to 1.8V from outside. We have limited amount of 3.3V and 1.8V signals and in order to support TE0712 we would have to add voltage translator buffers for 3.3V to 1.8V translation which in my opinion is an overkill. 

In the evening I will post schematics  of the modules that show the problem in a clearer way. 

Anyway, supporting different modules from Trenz is cumbersome.

zdun8, 2020-11-24:
The issue can be summarized to the following statement: there are not enough 1.8 V IOs on TE0712 to support it. 

I have compared TE0712, TE0720 and TE0715-15/30 in this spreadsheet https://docs.google.com/spreadsheets/d/1bnEOR_nFoHTlCReCy4KJ_HS_cJoHPDwn3MEk9fibMuM/edit?usp=sharing

Bank 14 which is connected to JB2 is 3.3 V fixed. If it was connected to JB2 pins as in other modules everything would work perfectly. Unfortunately there are not enough 1.8 V to move them to other bank/connector. 

[image]

[image]

gkasprow, 2020-12-16:
OK, let's leave only Zynq support then.

closed 2021-06-16 by gkasprow