sinara-hw/Fast_Servo #17: [sch review] C22 connections

filipswit, 2020-12-12:
In current configuration pins 3, 4, 6, 8 and 1, 3, 5, 7 are tied together. I think you should rename net labels for 5, 6, 7, 8.

[image]
[image]

filipswit, 2020-12-13:
According to #12, it is in recommended design, but It still shows unconnected nets. My proposition is to add under dash suffix to eliminate it.

[image]

zdun8, 2020-12-20:
Electrically it's still the same net as they are shorted internally in the ADC, so it would be a confusing. I will add a note. 
I've added a rule to not generate DRC for this signals not being connected, but I don't know if there is a way not to show the unconnected nets in the PCB. 

I will change the schematics as follows so it will not generate DRC and will still be clear. 
[image]

closed 2021-06-16 by gkasprow