sinara-hw/Fast_Servo #45: [pcb review] Split plane on L2 and L5

filipswit, 2020-12-14:
What for is this unconnected copper?

[image]

zdun8, 2021-01-02:
This is a recommended layout for the SI5340 oscillator. Please have a look on page 47 and 48 https://www.silabs.com/documents/public/reference-manuals/Si5341-40-D-RM.pdf

filipswit, 2021-01-02:
Sorry I placed to small screenshot. I'm thinking about this.
[image]

zdun8, 2021-01-02:
That's strange, this is exactly the same cutout as for the oscillator for SI5340, maybe it's a copy paste error or some Altium bug, I will remove this, thanks.

closed 2021-06-16 by gkasprow