sinara-hw/Fast_Servo #51: v1.0 debugging

gkasprow, 2022-04-03:
P3V3 is 3.2V instead of 3V3
PoE works
All rails work
Module with ZynQ board consumes 470mA without Silabs chip which still didn't arrive

gkasprow, 2022-04-03:
JTAG works, but XADC monitor shows 0

gkasprow, 2022-04-03:
XADC works after loading pre-built image from Trenz

jmatyas, 2022-08-05:
- Front Panel LEDs work
- Ethernet LEDs work
- LED1, LED2, LED3 works
- Si5340 configuration and programming via I2C successful ( #54 );  Si5340_nRST OK
- Clock_100M from OSC1 works
- UART works

### ADC LTC2195
- [x] SPI communication
- [ ] Data path
- [x] DCO path

### DAC AD9117
- [ ] SPI communication
- [x] Data Path (signal generation on both channels)

jmatyas, 2022-09-21:
Ethernet connection established

jmatyas, 2022-09-27:
ADC AUX AD7266
- [x] SPI communication OK
- [x] input address lines OK
- [x] range and SGL/!DIFF OK
- [x] sampled values OK

DAC AUX AD5722
- [x] SPI communication OK (both read and write)
- [x] BIN line OK
- [x] nCLR line OK
- [x] output values OK

jmatyas, 2022-10-27:
- microSD card slot and booting OK
- EEM I2C from EEM connector 
  - both LM75 OK
  - EEPROM OK
- GPIO header OK
- LVDS header OK

closed 2024-06-14 by kaolpr