sinara-hw/Fast_Servo #57: Differential pairs inversion

jmatyas, 2023-05-29:
A few of the differential lines are inverted:
  * FPGA_CLK1
  * FPGA_CLK2 (already marked on schematics)
  * EEM LVDS1
  * EEM LVDS2
  * EEM LVDS3
  * EEM LVDS4
  * EEM LVDS5
  * EEM LVDS7

The following lines  are marked as inverted, even though they aren't:
  * OUT1A
  * OUT1B
  * OUT1C

Those were found in addition to the ones reported in issue [#55](https://github.com/sinara-hw/Fast_Servo/issues/55#issue-1362053899)

maciejprzybysz, 2024-11-12:
As far as I understand this issue could be handled by changing pin polarity on Zynq. Is there anything to be fixed in hardware? @sbourdeauducq @jmatyas

linuswck, 2024-11-13:
Yes. Changing the pin polarity on Zynq can resolve this issue. As far as I know, no extra hardware change is needed.

maciejprzybysz, 2024-11-13:
Notes updated, due to this pinout tables:
https://shop.trenz-electronic.de/Download/?path=Trenz_Electronic/Pinout

closed 2024-11-13 by maciejprzybysz