sinara-hw/Fast_Servo #62: DAC Output Discrepency between Channels

linuswck, 2024-11-15:
# Description
DAC Output 0 and DAC Output 1 should have the same output voltage range but it is not. I suspect there is something wrong with the DAC AFE Stage.

The prototype board we received has the ADA4930-2YCPZ used as a differential to single end converter but the differential to single end operation is not specified in the OpAmp datasheet. 
[image]

# Point of Measurement
Measurements are taken with Fluke 87V at the AD9117-2 output resistors and DAC SMA Output Port.
[image]
[image]

# Measurement Result
|     | AD9117-2 Output | DAC SMA Output | Unsigned Data |
| -------- | ------- | ------- | ------- |
| Channel 0 | -427.8mV   | -157mV | 0x0000 |
| Channel 1 | -426.1mV   |-163mV  |  0x0000 |
| Channel 0 | 429.2mV    | 258mV  | 0x3FFF|
| Channel 1 | 424.3mV    |323mV   |  0x3FFF |

Note: AD9117-2 DAC is set to received unsigned data before the measurement is taken.

gkasprow, 2024-11-15:
Every opamp is specified for diff to SE conversion.

mkiepiela, 2025-02-05:
This problem is caused by the DAC AFE oscillations pointed out in https://github.com/sinara-hw/Fast_Servo/issues/64 and after fixing this issue, the output values are like:

|     | AD9117-2 Output | DAC SMA Output | Unsigned Data |
| -------- | ------- | ------- | ------- |
| Channel 0 | -424,27 mV  | -271.54 mV | 0x0000 |
| Channel 1 | -422,87 mV  |-272.62 mV  |  0x0000 |
| Channel 0 | -0,06 mV   | 31.19 mV  | 0x2000|
| Channel 1 | 0,02 mV   |29.15 mV   |  0x2000 |
| Channel 0 | 424,4 mV   | 333.60 mV  | 0x3FFF|
| Channel 1 | 422,9 mV   |330.67 mV   |  0x3FFF |

Now the discrepancy between the channels seems to be correct and acceptable.

Note: DC measurements taken with Uni-T UT181A.

---
[image]