sinara-hw/Fast_Servo #64: DAC Output Noise Coupling

linuswck, 2024-11-15:
# Description
When I connect the DAC Output 0 to an oscilloscope and I always see a high frequency sin wave being coupled to the DAC output SMA port. This high frequency sin wave exists even when the fast servo is powered up without the SD Card installed (so nothing is initialized). I am not sure if a board issue or a measurement problem.
[image]

# Connection Topology
Fast Servo DAC Output Port 0 -> SMA - to - SMA Cable -> SMA - to - BNC Adapter -> Oscilloscope BNC Port

# Other Info
Fast servo is installed inside a crate.
I tried powering it via PoE and power jack. The output noise makes no differences.
I have tested GND continuity between fast servo's and the SMA cable's, adapter's GND.
Oscilloscope Input Bandwidth is set to Full. This sin wave is still visible even after setting bandwidth to 20MHz.

mkiepiela, 2025-02-05:
Oscillations in the outputs come from an unstable amplifier in the DAC AFE.

This is caused by the parasitic inductance of the feedback capacitor and PCB tracks, as well as the parasitic capacitance in the summing node (amplifier inputs). This creates a filter pole and leads to excitation of the circuit.

When the feedback capacitors are removed, the oscillations disappear.
When they are mounted, the oscillation frequency is about 770-790 MHz (depends on the channel).

[image]

Experimentally, I added inductance to the feedback capacitors (soldering them on pieces of wire), expecting this to increase the problem (lowering the oscillation frequency).

[image]

The oscillation frequency decreased from 790 MHz to 600 MHz (green channel in the figures).

[image]

I'm going to test it with low ESL RF capacitors and update this topic.

Recommendations:
- use of RF capacitors with low ESL, preferably in smaller cases
- improved PCB layout (shorter and wider tracks and GND cutouts under the capacitor pads)

I am not closing this topic until the recommendations are implemented in the new revision.

---
[image]

linuswck, 2025-02-19:
I am testing the DAC AFE circuit using the EVAL-FDA-2 evaluation board to verify whether PCB layout has any contribution to the oscillation. The ADA4930-2 outputs oscillate even with RF capacitor being installed.

# Test Setup
RF Capacitor: 33pF (P/N: 06031U330FAT2A)
All Resistors used are 0402 0.1% tol.
Rf: 499ohm
Rg: 348ohm

Both IN-, IN+ are tied to GND via a 20ohm resistor
Both OUT-, OUT+ has a 0R series resistor installed.
Both Vocm are tied to GND.

+2.5V Power Source: TPS7A4901
-2.5V Power Source:  TPS7A3001

PD_n pins are tied to VS+.

Waveform are measured at OUT2+.

# Waveform
OUT1+ and OUT2+ have the same shape of waveform.
[image]
(Note: The scope used only has 500MHz bandwidth.)

linuswck, 2025-02-20:
OpAmp can be stabilized by adding an input capacitor across opamp inputs. But, the minimum value required depends on the PCB layout.
[image]

# Test Setup on  EVAL-FDA-2
This is tested on the EVAL-FDA-2 board. Same setup as previous reply. Both inputs are GND-ed via a 20ohm resistor. 
I tested the following values of capacitors.
| P/N | Value | Std deviation| Stable or Not? |
| -----| ------ | --------------------- | ---------------- |
| KGQ05FCG1H120FH | 0402 12pF +-1% | 7.61mV | Unstable |
| KGQ05FCG1H200FH | 0402 20pF +-1% | 6.25mV | Unstable |
| 06031U330FAT2A | 0603 33pF +-1% | 242.9uV | Stable |
| 06031U470JAT2A | 0603 47pF +-1% | 245.7uV | Stable |
| GRM188R71H221KA01D | 0603 220pF +-5% | 249.5uV | Stable |
(Note: Only 220pF capacitor is not RF capacitor)

### Waveforms
Pink Waveform is connected to OUT2+, which does not have 33pF feedback capacitor installed.
Green Waveform is connected to OUT1+, which is the DUT.

#### 0402 20pF +-1%
[image]

#### 0603 220pF+-1%
[image]

# Test Setup on the Fast-Servo
| P/N | Value | Std deviation | Stable or Not? |
| -----| ------ | --------------------- | ---------------- |
| 06031U330FAT2A | 0603 33pF +-1% | 3.51mV | Unstable |
| 06031U470JAT2A | 0603 47pF +-1% | 3.99mV | Unstable |
| 06031U680FAT2A | 0603 68pF +-1% | 1.31mV | Unstable |
| 06031U330FAT2A + 06031U680FAT2A | 0603 33pF + 68pF +-1% | 523uV | Stable |
| 06031U330FAT2A + 06031U680FAT2A | 0603 68pF + 33pF + 68pF +-1% | 503uV | Stable |
Note: Test Results are done on DAC1 Output

### Waveforms
Green Waveform is connected to Fast-servo Outputs.

#### 0603 33pF +-1%
[image]

#### 0603 68pF + 33pF + 68pF +-1% DAC1 Output
[image]

# Discussion
The added input capacitor shapes the noise gain of the FDA as discussed by this [app note from Ti] (https://www.ti.com/lit/an/slyt733/slyt733.pdf?ts=1740041287236&ref_url=https%253A%252F%252Fwww.google.com%252F). 
(Other useful forum post: https://e2e.ti.com/support/amplifiers-group/amplifiers/f/amplifiers-forum/664978/why-is-this-differential-mfb-lowpass-filter-oscillating)

As suggested by the ADA4930 datasheet(Layout Grounding and Bypassing section), Stray transmission line capacitance + package capacitance does contributes to the oscillation but as suggested by the EVAL-FDA-2 test result, a input capacitor is needed to stabilize the opamp. I think we should use 0402 resistor and (RF) capacitor for the FDA feedback path.

I think the DAC output is still very noisy compared to the DAC output range and 14 bit resolution. Should we build MFB differential low pass filter or add some kinds of filtering behind the DAC output?

linuswck, 2025-02-20:
Here is the output noise spectrum of DAC OUT1 on Fast Servo
Equipment: Keysight N9020A Signal Analyzer

0-1MHz Range
[image]

0-50MHz Range
[image]

0-500MHz Range
[image]

linuswck, 2025-02-21:
@gkasprow any thoughts?

linuswck, 2025-03-04:
> Maybe add a few ohm isolation resistors. 5Ohm or so

I think that is more likely a cable related issue. That's some unknown brand-ed cable. I tried cable that is fully specified. It works fine.

linuswck, 2025-03-04:
@gkasprow 
I know that ADA4930 should be one of the ADI FDA with lowest noise. But, I think its noise spec is only achievable when its differential output is used differentially? Particularly, the common mode noise is high as we can see from the measurement I took with the official evaluation board earlier.

[image]
C4: OUT1_P | C1: OUT1_N

I took the measurement again on the evaluation board with the same setup. We do not have any high bandwidth differential probe so the best I can do is to minus the both channels' readings digitally on the scope. But, we can still observe that generally OUTP and OUTN exhibits a significant amount common noise.