sinara-hw/Fast_Servo #65: Incorrect DAC AFE Output Range

linuswck, 2025-02-05:
From #5, the DAC AFE Output Range is expected to be +-1V (Differential output) or +- 0.5V (Single-Ended output)

However, AD9117-2 output is Unipolar: 0 - IoutFS. (Configured as 0 - 20mA) and  the simulation assumed the AD9117-2 to be bipolar output with its output range as -20mA - 20mA. This causes the output range to be halved.

In other words, the current DAC AFE Output Range is +-0.5V(Differential output) or +- 0.25V (Single-Ended output)

mkiepiela, 2025-02-13:
The ~30 mV offset in output voltage range is due to the type of amplifier used in the AFE.

ADA4930 is differential output amplifier optimized for driving ADC with 0.9V common mode voltage.
The output common mode voltage is set by Vocm input (in FastServo tied to GND).

The offset voltage between Vocm pin and actual output common mode voltage is specified in datasheet for Vocm = 0.9V (assuming single +5 V supply).

[image]

1. In FastServo, only one (positive) of the differential outputs is used as a single ended output reffered to ground. 
This means that the Vocm offset adds to the output signal and is a source of asymmetry in the output voltage range.

2. Since ADA4930 is powered by +/-2.5 V and the Vocm pin is tied to GND, it is 2.5 V above the negative supply voltage. This is even outside the range specified in the datasheet (0.5 - 2.3 V above the negative supply). For this reason, the observed common mode offset (effectively a shift in the output signal range) is much higher than that specified for Vocm = 0.9 V (-25 - +15 mV, typically +0.35 mV).

As an exercise, I changed the positive and negative supply voltage values to move Vocm closer to the negative supply voltage.
I then measured the output signal (Vocm offset) for the DACs set at mid-range (0x2000, expected output signal value of 0 V). Vocm offset decreased as expected.

| Vs- [V] | Vs+ [V] | Vocm (for both DAC channels) |
|---|---|---|
| -2.5 | +2.5 | 28-30 mV |
| -2 | +3 | ~20 mV |
| -1.22 | +3 | 15-18 mV |
| -1.22 | +3.73 | 7-10 mV |

linuswck, 2025-02-20:
I think 28mV offset voltage is a little bit too much for such small output range. AD9117-2 Dac actually has a AUXDAC (one for each channel) that we can use. Or we can use a resistor divider to trim the offset?

linuswck, 2025-02-20:
I think the original designer assumed the OUTP and OUTN of the DAC could generate -ve output voltage but AD9117-2 output is [unipolar](https://ez.analog.com/data_converters/high-speed_dacs/f/q-a/22556/ad9116/167562). 

[image]

On fast-servo at OUT_P / OUT_N pin, using up the full 0-20mA current range, you can only generate a sin wave centered at 200mV with 200mV amplitude (200mV +-200mV), which is halved the range on what the simulation suggested. This can explain why the output is about halved.
Note: The extra small gain is contributed by the input impedance of the Dac AFE circuit, which can be eliminated by setting a Dac registers.