sinara-hw/Fast_Servo #69: Fast_Servo - v2.0 - fix list & discussion

mkiepiela, 2025-02-19:
We are going to make a new revision of Fast_Servo in Technosystem. Issues to fix:

- [ ] https://github.com/sinara-hw/Fast_Servo/issues/68
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/65
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/64
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/63
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/62
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/67
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/66
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/61
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/58
- [ ] https://github.com/sinara-hw/Fast_Servo/issues/53

Feel free to add any suggestions/issues/ideas for new revision.

gkasprow, 2025-09-18:
I'd add optional (DNP) LC filter between DAC and LT1819 and also optional 0R instead of LT1568 for users who want higher BW than just 3MHz

kaolpr, 2025-09-18:
So basically we implement @linuswck design + what @gkasprow suggested. I'll see when we can schedule work on V2.